zhaojh329/oui #1: element-ui vs iview?

zhaojh329, 2017-12-12:
element-ui vs iview?

zhaojh329, 2017-12-17:
I have switched to iView.

closed 2017-12-20 by zhaojh329

zhaojh329, 2018-04-10:
feeling.

imudin, 2018-04-10:
@zhaojh329 haha I see. The same here my feeling is taking me to iView too. But I am doing company's project. If it is not stable enough, It is all on me. So having hard time to choose one of them. Personally I love iView, so pretty.  but element-ui has more stars, and it has good example vue-element-admin.

imudin, 2018-04-10:
What about IE support?

zhaojh329, 2018-04-10:
Iview has also been used by many well-known companies.

zhaojh329, 2018-04-10:
https://www.iviewui.com/docs/guide/introduce-en#Who's_using_iView

imudin, 2018-04-10:
I see! thank you

gustawdaniel, 2018-10-11:
In my opinion eleme.io has problem with screen scaling. Especially lacking support for mobile devices is distressing:

> https://github.com/ElemeFE/element/issues/3656

Although community asked about support for mobile in last comments in issue:

> https://github.com/ElemeFE/element/issues/7237

it was ignored.

I switched from eleme.io to iview today.