zhaojh329/oui #3: [Docs] Setting up OpenWRT build system

arky, 2018-08-31:
Can you please add instructions for setting up OpenWRT build system in contributors document. I think this will make it easier for new contributors to setup the environment.

zhaojh329, 2018-09-02:
I will write the document after the basic framework is completed.

closed 2019-05-20 by zhaojh329

acimanx, 2019-09-27:
@zhaojh329
Please add instructions to make it beginner-friendly 
🙌🙌🙌

zhaojh329, 2019-09-27:
https://zhaojh329.github.io/oui/

acimanx, 2019-09-27:
> https://zhaojh329.github.io/oui/
Thanks for your fast reply, but I've read it before.
There are things that I cannot understand. 

For example:
1. Where should I execute 
`echo "src-git oui https://github.com/zhaojh329/oui.git" >> feeds.conf.default`
in my openwrt home directory? Or?

2. I don't have a scripts directory to execute:
`./scripts/feeds update oui`
[url=https://imgbb.com/][img]https://i.ibb.co/M57Lbzk/1.png[/img][/url]

And etc...