zhaojh329/oui #19: yorkie fails with errno 134

aparcar, 2019-07-03:
```
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core:oui-ui-core -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/a/src/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/a/src/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/include -I/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include/fortify -I/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core:oui-ui-core -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/a/src/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/a/src/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/include -I/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include/fortify -I/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include " LDFLAGS="-L/home/a/src/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/a/src/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/a/src/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro " make  -C /home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core:oui-ui-core -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core'
npm install

> yorkie@2.0.0 install /home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core/node_modules/yorkie
> node bin/install.js

setting up Git hooks
can't find .git directory, skipping Git hooks installation
Aborted
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 134
npm ERR! yorkie@2.0.0 install: `node bin/install.js`
npm ERR! Exit status 134
npm ERR! 
npm ERR! Failed at the yorkie@2.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/a/.npm/_logs/2019-07-03T11_18_05_215Z-debug.log
make[4]: *** [Makefile:4: .tmp/.npm_stamp] Aborted
make[4]: Leaving directory '/home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core'
make[3]: *** [Makefile:69: /home/a/src/openwrt/build_dir/target-mips_24kc_musl/oui-ui-core/.built] Error 2
make[3]: Leaving directory '/home/a/src/openwrt/feeds/oui/oui-ui-core'
time: package/feeds/oui/oui-ui-core/compile#11.83#0.75#9.94
make[2]: *** [package/Makefile:109: package/feeds/oui/oui-ui-core/compile] Error 2
make[2]: Leaving directory '/home/a/src/openwrt'
make[1]: *** [package/Makefile:103: /home/a/src/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/a/src/openwrt'
make: *** [/home/a/src/openwrt/include/toplevel.mk:218: world] Error 2

```

aparcar, 2019-07-03:
If I can help with any additional information please let me know!

zhaojh329, 2019-07-03:
What is the version of your node?

zhaojh329, 2019-07-04:
Please update feed and see https://zhaojh329.github.io/oui/guide/compile.html

aparcar, 2019-07-04:
Works now, thank you very much for the fixup! I think it should use the OpenWrt npm version by default and it should be a host dependency. For that very reason the whole toolchain of OpenWrt's build root is compiled, to avoid errors like that.

closed 2019-07-04 by aparcar