zhaojh329/oui #21: Add common use cases

aparcar, 2019-07-04:
While OUI is a nice web interface, it is basically a rewrite of the existing Luci interface with all its strengths **and** weaknesses. Meaning, regular users are (instantly) overwhelmed by the amount of options and number of required steps to perform *simple* tasks, like creating a guest WiFi with limited bandwidth.

It would be fantastic if you could add a simplified version which offers a limited granularity of options while performing complex configurations at once. A number of examples is found [here](https://forum.openwrt.org/t/common-use-case-for-simplified-web-interface/38866/).

I'd be happy to collaborate on the implementation, as I see the urgent need for a simple (and optionally advanced) web interface for OpenWrt.

zhaojh329, 2019-07-04:
Oui is still under development, there are many features that need to be
developed, and there are many things that need to be improved.

closed 2020-02-11 by zhaojh329

aparcar, 2020-02-12:
:man_shrugging: