zhaojh329/oui #23: [Build] package/feeds/oui/oui-ui-core/compile compile error

belowSt, 2019-07-16:
Unable to complete error. It seems there is an error in oui-bwm

`cho "libc" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.16#0.03#0.20
make[2]: Entering directory '/media/data/Projekte/OpenWRT/openwrt/package/libs/libjson-c'
echo "libjson-c" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libjson-c.default.install
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/package/libs/libjson-c'
time: package/libs/libjson-c/compile#0.21#0.05#0.30
make[2]: Entering directory '/media/data/Projekte/OpenWRT/openwrt/package/utils/lua'
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/package/utils/lua'
time: package/utils/lua/compile#0.20#0.03#0.26
make[2]: Entering directory '/media/data/Projekte/OpenWRT/openwrt/package/libs/libubox'
echo "libubox" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
echo "libblobmsg-json" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
echo "jshn" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
echo "libjson-script" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/package/libs/libubox'
time: package/libs/libubox/compile#0.61#0.20#1.06
make[2]: Entering directory '/media/data/Projekte/OpenWRT/openwrt/package/system/ubus'
echo "libubus" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/ubus.default.install
echo "ubus" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/ubus.default.install
echo "ubusd" >> /media/data/Projekte/OpenWRT/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/ubus.default.install
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/package/system/ubus'
time: package/system/ubus/compile#0.49#0.12#0.73
make[2]: Entering directory '/media/data/Projekte/OpenWRT/openwrt/feeds/oui/oui-io-util'
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/feeds/oui/oui-io-util'
time: package/feeds/oui/oui-io-util/compile#0.14#0.03#0.21
make[2]: Entering directory '/media/data/Projekte/OpenWRT/openwrt/feeds/oui/oui-bwm'
make[2]: *** No rule to make target '/media/data/Projekte/OpenWRT/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-4.14.132/.config', needed by '/media/data/Projekte/OpenWRT/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/oui-bwm/.built'.  Stop.
make[2]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt/feeds/oui/oui-bwm'
time: package/feeds/oui/oui-bwm/compile#0.14#0.03#0.21
make[1]: *** [package/Makefile:109: package/feeds/oui/oui-bwm/compile] Error 2
make[1]: Leaving directory '/media/data/Projekte/OpenWRT/openwrt'
make: *** [/media/data/Projekte/OpenWRT/openwrt/include/toplevel.mk:218: package/feeds/oui/oui-ui-core/compile] Error 2
`

zhaojh329, 2019-07-16:
Please list your compilation environment and operation steps.

belowSt, 2019-07-16:
Sorry, you can close this issue. It was my fault (Some dependencies were missing)

closed 2019-07-16 by belowSt