zhaojh329/oui #40: compile error

p3rtinax, 2019-12-11:
Environment: MT7620 ( ramips ) - 18.06.5

Description:
I've got an error while compiling, lua-cjson package as oui-rpc-core dependency 
the package is at this URL : 
http://downloads.openwrt.org/releases/18.06.5/packages/mipsel_24kc/packages/lua-cjson_2.1.0-2_mipsel_24kc.ipk

how can I add this ipk manually to the building process ?

`Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for oui-rpc-core:
 *      lua-cjson
 * opkg_install_cmd: Cannot install package oui-rpc-core.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/user/openwrt/openwrt-18.06.5'
package/Makefile:108: recipe for target '/home/user/openwrt/openwrt-18.06.5/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/user/openwrt/openwrt-18.06.5/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/user/openwrt/openwrt-18.06.5'
/home/user/openwrt/openwrt-18.06.5/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
user@ubu18-server:~/openwrt/openwrt-18.06.5$ `

Ai3008, 2019-12-16:
hello,
Have you find a solution for this issue please, i have the exact same since 2 days now and can't fix it no matter what i tried.
Thank you by advance.

zhaojh329, 2019-12-16:
This question does not belong to oui.

Ai3008, 2019-12-17:
Is it related to openwrt system?? i am trying to install the Oui library since 5 days now and i still get this lua-cjson error even if i have the lua-cjson package installed on my machine, i tried multiple solutions but none worked, can you suggest something to solve this please?

p3rtinax, 2019-12-18:
@Ai3008 
No :(

@zhaojh329 
how this is not related to OUI?
 the error says it can not find dependencies for oui-rpc-core : lua-cjson

any solution?

Ai3008, 2019-12-19:
@p3rtinax I solved the problem!!
if you are using an openwrt/Lede system it can be the issue, cause i switched for an openwrt 18.06.2 version yesterday and it compiled very well, maybe we had an old version of the OS that doesn't support the dependencies? i can't say
I hope this can help you too :)

closed 2020-02-11 by zhaojh329