zhaojh329/oui #41: OUI Framework - Encountered error

mbabon, 2019-12-16:
Environment: (put here arch, model, OpenWrt version)

Description:
We cloned the master branch of OUI framework and run the project but we get this error 
**"Proxy error : Could not proxy request /ubus from 127.0.0.0.1:8080 to http://127.0.0.1 (ECONNREFUSED)."** 

We tried to install ubus in our local but still get the same error. 
We don't know if we need to set up a server or OpenWrt to run the project in our local.

Hope someone can give us an input. Thanks

[image]
[image]

zhaojh329, 2019-12-16:
https://zhaojh329.github.io/oui/guide/

m-bbn, 2019-12-17:
> https://zhaojh329.github.io/oui/guide/

Hi @zhaojh329 , thank you for your response but we already follow that steps https://zhaojh329.github.io/oui/guide/getting-started.html#debugging-front-end-code

lqfcu3mtd, 2019-12-18:
I had the same problem，follow that steps has the same error. 
[image]
[image]
[image]

module.exports = {
  indexPath: 'oui.html',
  productionSourceMap: false,
  devServer: {
    proxy: {
      '/ubus': {
        target: '192.168.1.1'
      },
      '/cgi-bin/': {
        target: '192.168.1.1'
      },
      '/views/': {
        target: '192.168.1.1'
      },
      '/i18n/': {
        target: '192.168.1.1'
      }
    }
  },
  pluginOptions: {
    i18n: {
      locale: 'en',
      fallbackLocale: 'en',
      localeDir: 'locales',
      enableInSFC: false
    }
  }
}

lqfcu3mtd, 2019-12-18:
In vue.config.js  the 'target' must contain 'http: //'   
`devServer: {
    proxy: {
      '/ubus': {
        target: 'http://192.168.1.1'
      },
      '/cgi-bin/': {
        target: 'http://192.168.1.1'
      },
      '/views/': {
        target: 'http://192.168.1.1'
      },
      '/i18n/': {
        target: 'http://192.168.1.1'
      }
    }
  }`

closed 2020-02-11 by zhaojh329