zhaojh329/oui #50: oui-ui-core: why not use cgi-io upstream package

feckert, 2020-04-08:
Environment: x86_64, APU3, OpenWrt master

Description:
@zhaojh329 Is there a reason why you have forked cgi-io into oui-io-util?
I think it is better to use the upstream project here and not just make a fork.
Of course there are advantages and disadvantages but I think the advantages outweigh the disadvantages here. The Upstream project is much further along and has already fixed bugs and added new features.

If this is OK I will prepare a pull request to remove this and use the upstream project from openwrt core. I do have to to some additional test on my environment.

The following commits are not added.

- So do we need them?
- Or are they already applied upstream?

https://github.com/zhaojh329/oui/commit/c7131d5c4ca748a192e9270b7de6bd3c7d5c276a#diff-81c7acce021b0b593a0b1aec6d2f5a0c
https://github.com/zhaojh329/oui/commit/28e26319514a1ab4f5383c8c3006b504b242e88e#diff-81c7acce021b0b593a0b1aec6d2f5a0c

zhaojh329, 2020-04-08:
The reason is that I hope oui can be used on older OpenWrt.

feckert, 2020-04-08:
This package is included in the following branches
- master
- openwrt-19.07 (current stable)
- openwrt-18.06 (EOL)
- lede-17.01 (EOL)
- for-15.05 (EOL)

So the only branch that is not supported is **for-14.07** which is also EOL.
So correct me if I'm wrong but I think the advantage for building on older OpenWrt is not  given.

zhaojh329, 2020-04-08:
https://github.com/openwrt/packages/pull/9105/files

zhaojh329, 2020-04-08:
Maybe I don't need to consider the old version of openwrt.

feckert, 2020-04-08:
> Maybe I don't need to consider the old version of openwrt.

When someone develops (me as well) in openwrt, he actually always takes the master or the latest stable (for now openwrt-19.07) for development. A new feature is actually never added to the EOL branches only bugfixes. If you want to support old version. Then you can make branches and change the API to  cgi-download vs. cgi-backup.

I have compiled with the upstream cgi-io and changed vue calles. After my tests tomorrow I will open a pullrequest with my requested changes. Thanks for you feedback :-)

By the way nice work :+1:

feckert, 2020-04-09:
I have added a pullrequest https://github.com/zhaojh329/oui/pull/52 for this.

closed 2020-04-09 by zhaojh329