zhaojh329/oui #54: I can't compile

castillofrancodamian, 2020-05-10:
Environment: OpenWrt Snapshots

Description: I follow the compilation instructions but it gives me error.

```
franco@GA-B150M-DS3H:~/openwrt$
make[2]: Entering directory '/home/franco/openwrt/scripts/config'
conf.c: In function 'main':
conf.c:620:6: warning: format not a string literal and no format arguments [-Wformat-security]
  620 |      _("\n*** The configuration requires explicit update.\n\n"));
      |      ^
conf.c:674:4: warning: format not a string literal and no format arguments [-Wformat-security]
  674 |    fprintf(stderr, _("\n*** Error during writing of the configuration.\n\n"));
      |    ^~~~~~~
conf.c:678:4: warning: format not a string literal and no format arguments [-Wformat-security]
  678 |    fprintf(stderr, _("\n*** Error during update of the configuration.\n\n"));
      |    ^~~~~~~
conf.c:689:4: warning: format not a string literal and no format arguments [-Wformat-security]
  689 |    fprintf(stderr, _("\n*** Error during writing of the configuration.\n\n"));
      |    ^~~~~~~
make[2]: Leaving directory '/home/franco/openwrt/scripts/config'
time: target/linux/prereq#0.53#0.51#1.26
make[1]: Entering directory '/home/franco/openwrt'
make[2]: Entering directory '/home/franco/openwrt/feeds/oui/oui-bwm'
touch /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/.prepared_69d9b7811950b3624bfeb921fe6c58a9_6664517399ebbbc92a37c5bb081b5c53_check
install -d -m0755 /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm
cp -fpR ./src/. /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm
touch /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/.prepared_69d9b7811950b3624bfeb921fe6c58a9_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/.configured_*
rm -f /home/franco/openwrt/staging_dir/target-mips_24kc_musl/stamp/.oui-bwm_installed
(cd /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/./; if [ -x ./configure ]; then find /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/ -name config.guess | xargs -r chmod u+w; find /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/ -name config.guess | xargs -r -n1 cp --remove-destination /home/franco/openwrt/scripts/config.guess; find /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/ -name config.sub | xargs -r chmod u+w; find /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/ -name config.sub | xargs -r -n1 cp --remove-destination /home/franco/openwrt/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -ffile-prefix-map=/home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm=oui-bwm -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm=oui-bwm -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm=oui-bwm -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/franco/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/franco/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/usr/include -I/home/franco/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/fortify -I/home/franco/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include " LDFLAGS="-L/home/franco/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/franco/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/franco/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/usr/lib -L/home/franco/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/.configured_68b329da9893e34099c7d8ad5cb9c940
make[2]: *** No rule to make target '/home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-4.19.115/.config', needed by '/home/franco/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/oui-bwm/.built'.  Stop.
make[2]: Leaving directory '/home/franco/openwrt/feeds/oui/oui-bwm'
time: package/feeds/oui/oui-bwm/compile#0.12#0.67#1.81
make[1]: *** [package/Makefile:113: package/feeds/oui/oui-bwm/compile] Error 2
make[1]: Leaving directory '/home/franco/openwrt'
make: *** [/home/franco/openwrt/include/toplevel.mk:220: package/feeds/oui/oui-ui-core/compile] Error 2
```

PyMario, 2020-07-29:
I got the same **ERROR** as you!

```bash
mario@mario:~/Desktop/openwrt$ make package/feeds/oui/oui-ui-core/compile V=s
time: target/linux/prereq#0.21#0.01#0.23
make[1]: Entering directory '/home/mario/Desktop/openwrt'
make[2]: Entering directory '/home/mario/Desktop/openwrt/feeds/oui/oui-bwm'
touch /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/.prepared_7c9d27424d14f1a56a22159c5cc9f3b8_6664517399ebbbc92a37c5bb081b5c53_check
install -d -m0755 /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm
cp -fpR ./src/. /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm
touch /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/.prepared_7c9d27424d14f1a56a22159c5cc9f3b8_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/.configured_*
rm -f /home/mario/Desktop/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.oui-bwm_installed
(cd /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/./; if [ -x ./configure ]; then find /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/ -name config.guess | xargs -r chmod u+w; find /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/ -name config.guess | xargs -r -n1 cp --remove-destination /home/mario/Desktop/openwrt/scripts/config.guess; find /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/ -name config.sub | xargs -r chmod u+w; find /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/ -name config.sub | xargs -r -n1 cp --remove-destination /home/mario/Desktop/openwrt/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -ffile-prefix-map=/home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm=oui-bwm -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm=oui-bwm -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm=oui-bwm -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/mario/Desktop/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/mario/Desktop/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/mario/Desktop/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/mario/Desktop/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include " LDFLAGS="-L/home/mario/Desktop/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/mario/Desktop/openwrt/staging_dir/target-mipsel_24kc_musl/lib -L/home/mario/Desktop/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/mario/Desktop/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=mipsel-openwrt-linux --host=mipsel-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/.configured_68b329da9893e34099c7d8ad5cb9c940
make[2]: *** No rule to make target '/home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.52/.config', needed by '/home/mario/Desktop/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/oui-bwm/.built'.  Stop.
make[2]: Leaving directory '/home/mario/Desktop/openwrt/feeds/oui/oui-bwm'
time: package/feeds/oui/oui-bwm/compile#0.10#0.02#0.11
make[1]: *** [package/Makefile:113: package/feeds/oui/oui-bwm/compile] Error 2
make[1]: Leaving directory '/home/mario/Desktop/openwrt'
make: *** [/home/mario/Desktop/openwrt/include/toplevel.mk:235: package/feeds/oui/oui-ui-core/compile] Error 2
```

zhaojh329, 2020-07-31:
Please compile openwrt completely first.

make -j 4

closed 2020-07-31 by zhaojh329

castillofrancodamian, 2020-08-03:
How do I compile it completely first?

SnailJason, 2020-08-14:
> How do I compile it completely first? Things like if I should select Luci in `make menuconfig` are not said.
Is it solved?

castillofrancodamian, 2020-08-18:
> > How do I compile it completely first? Things like if I should select Luci in `make menuconfig` are not said.
> > Is it solved?

I still don't know how I have to compile with `make menuconfig`.

zhaojh329, 2020-08-18:
> > > How do I compile it completely first? Things like if I should select Luci in `make menuconfig` are not said.
> > > Is it solved?
> 
> I still don't know how I have to compile with `make menuconfig`.

https://zhaojh329.github.io/oui/guide/compile.html

castillofrancodamian, 2020-08-18:
> > > > How do I compile it completely first? Things like if I should select Luci in `make menuconfig` are not said.
> > > > Is it solved?
> > 
> > 
> > I still don't know how I have to compile with `make menuconfig`.
> 
> [zhaojh329.github.io/oui/guide/compile.html](https://zhaojh329.github.io/oui/guide/compile.html)

I just tried to compile but an error appeared and I still couldn't test it.