zhaojh329/oui #59: oui compile failed

xuchao49, 2020-06-23:
/data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/oui-bwm/main.c:13:
include/linux/sem.h:14:5: error: requested alignment is not an integer constant
In file included from include/linux/irq.h:353:0,
                 from /data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/linux-3.10.108/arch/x86/include/asm/hardirq.h:5,
                 from include/linux/hardirq.h:8,
                 from include/net/netprio_cgroup.h:17,
                 from include/linux/netdevice.h:48,
                 from include/uapi/linux/if_arp.h:26,
                 from include/linux/if_arp.h:27,
                 from include/net/arp.h:5,
                 from /data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/oui-bwm/main.c:13:
include/linux/irqdesc.h:77:1: error: requested alignment is not an integer constant
In file included from include/linux/hardirq.h:8:0,
                 from include/net/netprio_cgroup.h:17,
                 from include/linux/netdevice.h:48,
                 from include/uapi/linux/if_arp.h:26,
                 from include/linux/if_arp.h:27,
                 from include/net/arp.h:5,
                 from /data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/oui-bwm/main.c:13:
/data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/linux-3.10.108/arch/x86/include/asm/hardirq.h:36:1: error: requested alignment is not an integer constant
/data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/linux-3.10.108/arch/x86/include/asm/hardirq.h:38:1: error: requested alignment is not an integer constant
In file included from include/uapi/linux/if_arp.h:26:0,
                 from include/linux/if_arp.h:27,
                 from include/net/arp.h:5,
                 from /data/xuchao/7621/mt7621sdk/mtk-openwrt-cc-v3.6-0616/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/oui-bwm/main.c:13:
include/linux/netdevice.h:530:2: error: requested alignment is not an integer constant
include/linux/netdevice.h:548:1: error: requested alignment is not an integer constant
include/linux/netdevice.h: In function 'rps_record_sock_flow':
include/linux/netdevice.h:621:3: error: implicit declaration of function 'raw_smp_processor_id' [-Werror=implicit-function-declaration]
include/linux/netdevice.h: At top level:
include/linux/netdevice.h:648:1: error: requested alignment is not an integer constant
include/linux/netdevice.h:1214:9: error: requested alignment is not an integer constant
include/linux/netdevice.h:1816:9: error: requested alignment is not an integer constant
include/linux/netdevice.h:1842:1: error: requested alignment is not an integer constant

baidxi, 2020-07-01:
内核太老了。

xuchao49, 2020-07-06:
最低支持多少内核，可以部分移植吗，

closed 2020-08-03 by zhaojh329

SnailJason, 2020-08-14:
同问，最低支持内核版本是多少

baidxi, 2020-08-14:
> 同问，最低支持内核版本是多少

建议4.4以上的内核。lede 17.01以上的版本。