zhaojh329/oui #69: Add Oui to Openwrt/Packages

arky, 2020-08-01:
Please add Oui feed for inclusion in Openwrt/packages (https://github.com/openwrt/packages). This enables ease of deployment to wider openwrt devices.

tiagogaspar8, 2020-09-19:
I'd love this too, It's an amazing interface, very well designed.
I can help if needed

mgscreativa, 2020-10-07:
It would be great to have it as a general package!

zhaojh329, 2020-10-08:
It seems that it needs to be split into a single package to merge into openwrt / packages.

tiagogaspar8, 2020-10-08:
Can we do that?

zhaojh329, 2020-10-09:
The best way to join OpenWrt is as feeds, like Luci.

tiagogaspar8, 2020-10-19:
But will the OpenWRT team allow another feed? I believe they only accept the merge of packages into the existing feeds.
I didn't take a look but is it that complicated to separate oui into several packages and get them merged?

tiagogaspar8, 2020-11-14:
@zhaojh329 have you ever tried to merge this as a feed? Have you ever seen anyone do that? I have been reading the packages and makefiles anda I believe that they don't fit any of the folders, aside from uhttpd, wich they could argue that would only be another version, I believe this requires another folder just like LuCi, since this is a "replacement".

Also, is there any way to make it run on nginx?

zhaojh329, 2020-11-15:
> @zhaojh329 have you ever tried to merge this as a feed? Have you ever seen anyone do that? I have been reading the packages and makefiles anda I believe that they don't fit any of the folders, aside [...]

No plan to support nginx.

tiagogaspar8, 2020-11-16:
And have you ever tried to merge your feed to OpenWRT?

closed 2020-11-26 by zhaojh329