zhaojh329/oui #75: Lua dir missing

TautvydasD, 2020-10-08:
Environment: (put here arch, model, OpenWrt version)
OpenWrt 18.06.8

Description:

```
CMake Error at /home/studentas/Documents/openwrt-gpl-ipq/staging_dir/host/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Lua (missing: LUA_INCLUDE_DIR)
Call Stack (most recent call first):
  /home/studentas/Documents/openwrt-gpl-ipq/staging_dir/host/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /home/studentas/Documents/openwrt-gpl-ipq/staging_dir/host/share/cmake-3.11/Modules/FindLua.cmake:193 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:11 (find_package)

```

zhaojh329, 2020-10-08:
I tested in OpenWrt master.

TautvydasD, 2020-10-08:
Error in my OpenWrt master.

```
In file included from /home/studentas/Documents/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/uhttpd/uhttpd.h:30,
                 from /home/studentas/Documents/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oui-httpd-git-2020.282.09110-b5746de/main.c:32:
/home/studentas/Documents/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/uhttpd/connection.h:32:10: fatal error: http-parser/http_parser.h: No such file or directory
 #include "http-parser/http_parser.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

```

Any suggestions?

zhaojh329, 2020-10-08:
Please update your feeds.

TautvydasD, 2020-10-08:
Thx for the help on OpenWrt master, but do you have any suggestions for the older version?

zhaojh329, 2020-10-09:
Fixed. Pls update.

closed 2020-10-09 by zhaojh329

TautvydasD, 2020-10-09:
So, there were multiple problems, but the solutions I think were:

1. Some lines in ```./openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libuhttpd-nossl/libuhttpd-2.0.3/src/utils.c``` were missing ```%X```, so I changed them:
```
From:
ustream_printf(us, "\r\n", len);
To:
ustream_printf(us, "%X\r\n", len);
```
2. Updated libuhttpd Makefile from the url:
https://github.com/openwrt/packages/blob/master/libs/libuhttpd/Makefile

3. Changed required Lua version in cmake file:
- ./openwrt/feeds/oui/oui-httpd/src/CMakeLists.txt :
```
From Line 11:
find_package(Lua REQUIRED)

To Line11:
find_package(Lua51 REQUIRED)
```

zhaojh329, 2020-10-09:
Yes.