zhaojh329/oui #76: Invalid or unexpected token

TautvydasD, 2020-10-09:
Environment: (put here arch, model, OpenWrt version)
OpenWrt 18.06.8

Description: After uploading packages to system and running oui on browser.

```
chunk-vendors.84231e0e.js:1 Uncaught SyntaxError: Invalid or unexpected token
app.1eac7929.js:1 Uncaught SyntaxError: Invalid or unexpected token
```

TautvydasD, 2020-10-09:
On OpenWrt master running only oui. ``` http://localhost``` shows only content of folder ``` /www ```

zhaojh329, 2020-10-09:
Try chrome.

TautvydasD, 2020-10-09:
On Chrome

[image]

zhaojh329, 2020-10-09:
Please stop other http server, such as uhttpd. And then restart oui-httpd.

closed 2020-10-09 by TautvydasD