zhaojh329/oui #79: Getting Error Compiling with 19.07

kumart4, 2020-11-10:
I am getting following error when trying to compile with 19.07. Any insights as to how i can solve this.

CMake Error at /home/rounak/Workbench/staging_dir/host/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Libev (missing: LIBEV_LIBRARY LIBEV_INCLUDE_DIR)
Call Stack (most recent call first):
  /home/rounak/Workbench/staging_dir/host/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindLibev.cmake:25 (find_package_handle_standard_args)
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rounak/Workbench/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/oui-httpd-git-2020.301.59264-fcb1d74/CMakeFiles/CMakeOutput.log".
Makefile:48: recipe for target '/home/rounak/Workbench/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/oui-httpd-git-2020.301.59264-fcb1d74/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[2]: *** [/home/rounak/Workbench/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/oui-httpd-git-2020.301.59264-fcb1d74/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/home/rounak/Workbench/feeds/oui/oui-httpd'

kumart4, 2020-11-10:
This is resolved

closed 2020-11-10 by kumart4

hjt88123, 2020-11-11:
Same error, could you tell me how to resovle?

sensaihk, 2020-11-11:
Select libev in libraries and make sure libuhttpd is greater than 3.4. First Compile libev and then try compiling oui.

zhaojh329, 2020-11-11:
> Select libev in libraries and make sure libuhttpd is greater than 3.4. First Compile libev and then try compiling oui.
Only make sure libuhttpd is greater than 3.4.

kumart4, 2020-11-11:
First add and compile libev then libuhttpd 3.4 and then try OUI

hjt88123, 2020-11-13:
Resolved, many thanks.