zhaojh329/oui #82: web page not showing up

wanderernxa, 2020-12-26:
Environment: raspberry pi 4 model b, latest openwrt pull from git

Description:

Hi,
I followed the instructions, did a clean compile and it installed on my raspberry pi. But when I go to the url, it is a blank page. Nothing is showing.. Any help?

thanks

wanderernxa, 2021-01-01:
after some more investigating and keeping in mind that vuejs is new to me, I think the problem occurs when it cannot find the language (locale) setting...is the locale set in openwrt for OUI? Anyone paste an a sample of their config file where lang or locale is defined so get_lang function can work properly?

thanks

wanderernxa, 2021-01-01:
ok, I figured it out...I compiled and deployed as normal. I then went into /etc/config/oui file and change the option lang from auto to en....rebooted and it worked perfectly! Hopefully this will help others.

closed 2021-01-01 by wanderernxa

predators46, 2021-01-04:
> ok, I figured it out...I compiled and deployed as normal. I then went into /etc/config/oui file and change the option lang from auto to en....rebooted and it worked perfectly! Hopefully this will help others.

can you explain creating files and running the theme