zhaojh329/oui #86: Not able to see oui theme in openwrt build image

hemanth4ap, 2021-01-24:
Environment: mt7628, OpenWrt 19.07)

Description:
We are not able to get OUI theme in openwrt environment.  

We fallowed below steps mentioned in README.md .

1. Add feeds
2. Configure
3. Compile
4 Flashed image

And tried to access UI but it showing empty page. Could you please let us know is any extra steps required to get OUI theme.

scramatte, 2021-01-27:
I've got same issue.  It looks that OUI has depencies such rpcd  or ubus that are not installed properly. Moreover  it looks that content is gzipped and  uhttpd server doesn't handle it properly.

zhaojh329, 2021-01-28:
Please disable uhttpd. Oui dosen't use uhttpd or modify listen port. Oui use oui-httpd.
https://github.com/zhaojh329/oui/tree/master/oui-httpd

predators46, 2021-02-08:
> Please disable uhttpd. Oui dosen't use uhttpd or modify listen port. Oui use oui-httpd.
> https://github.com/zhaojh329/oui/tree/master/oui-httpd

same as i experienced i don't see oui to login. and I saw the url http://192.168.1.1/#/wizard

predators46, 2021-02-08:
i found another problem i already installed oui-app-wizard but i get below notification

` You have not installed the wizard page for OUI, you can either install the OUI built-in wizard page app(oui-app-wizard), or create your own wizard page app.  `

closed 2022-07-21 by zhaojh329