zhaojh329/oui #90: oui-httpd download.c static urldecode conflicting types with libuhttpd3.9

topeqin, 2021-02-02:
Environment: (arm, openwrt 19.07)

Description:

v19.07/build_dir/target-arm-buildroot-linux-gnueabi_glibc/oui-httpd-git-2021.028.43331-1ff4a9b/download.c:30:13: error: conflicting types for 'urldecode'
   30 | static bool urldecode(char *buf)
      |             ^~~~~~~~~
In file included from /home/gqin/openwrt/v19.07/staging_dir/target-arm-buildroot-linux-gnueabi_glibc/usr/include/uhttpd/uhttpd.h:35

topeqin, 2021-02-02:
Hello jianhui

I meet conflict static local funcion urldecode in download.c with openwrt feeds package libuhttpd whose version is 3.9
dl url https://github.com/zhaojh329/libuhttpd
if something steps miss then result this issue. if yes, please comments. I just add one fix in my local first. 

Thanks,
Qin Guang

closed 2021-02-02 by zhaojh329

predators46, 2021-02-08:
> Hello jianhui
>
> I meet conflict static local funcion urldecode in download.c with openwrt feeds package libuhttpd whose version is 3.9
> dl url https://github.com/zhaojh329/libuhttpd
> if something steps [...]

can you share the file libuhttpd3.9.0