zhaojh329/oui #93: build error due to missing urldecode from download.c

laoshaw, 2021-02-11:
master branch openwrt, failed to build oui as urldecode is not defined in the repo.

Description:

```
download.c:77:5: error: implicit declaration of function 'urldecode' [-Werror=implicit-function-declaration]
     urldecode(params->path, sizeof(params->path), params->path, strlen(params->path));
```

mengqingmeng, 2021-02-17:
> master branch openwrt, failed to build oui as urldecode is not defined in the repo.
>
> Description:
>
> ```
> download.c:77:5: error: implicit declaration of function 'urldecode' [...]

I had this problem too,how to solve this?

zhaojh329, 2021-02-19:
https://github.com/openwrt/packages/pull/14789

closed 2021-02-19 by zhaojh329