zhaojh329/oui #98: too many arguments to function 'uh_server_new'

JasonHonor, 2021-02-27:
Environment: (put here arch, model, OpenWrt version)

OpenWrt 19.07-master, mips32

Description:

```
make[5]: Entering directory '/home/runner/work/openwrt/openwrt/openwrt/build_dir/target-mips_24kc_musl/oui-httpd-git-2021.051.26375-ec29a24'
[ 40%] Building C object CMakeFiles/oui-httpd.dir/main.c.o
/home/runner/work/openwrt/openwrt/openwrt/build_dir/target-mips_24kc_musl/oui-httpd-git-2021.051.26375-ec29a24/main.c: In function 'main':
/home/runner/work/openwrt/openwrt/openwrt/build_dir/target-mips_24kc_musl/oui-httpd-git-2021.051.26375-ec29a24/main.c:146:11: error: too many arguments to function 'uh_server_new'
     srv = uh_server_new(loop, addr, port);
           ^~~~~~~~~~~~~
In file included from /home/runner/work/openwrt/openwrt/openwrt/build_dir/target-mips_24kc_musl/oui-httpd-git-2021.051.26375-ec29a24/main.c:32:
/home/runner/work/openwrt/openwrt/openwrt/staging_dir/target-mips_24kc_musl/usr/include/uhttpd/uhttpd.h:132:19: note: declared here
 struct uh_server *uh_server_new(struct ev_loop *loop);
                   ^~~~~~~~~~~~~
CMakeFiles/oui-httpd.dir/build.make:81: recipe for target 'CMakeFiles/oui-httpd.dir/main.c.o' failed
make[5]: *** [CMakeFiles/oui-httpd.dir/main.c.o] Error 1

```

zhaojh329, 2021-02-27:
Pls update oui.
```
./scripts/feeds update oui
```

JasonHonor, 2021-02-28:
Thanks, I had updated oui by the following command:

```
./scripts/feeds update -a

```

and the commit version of oui-httpd-git  was already ec29a24, how next?

zhaojh329, 2021-02-28:
```
rm -rf feeds/oui*
./scripts/feeds update oui
./scripts/feeds install -a -p oui
```

zhaojh329, 2022-07-21:
Vue3 + Nginx + Lua is coming...

closed 2022-07-21 by zhaojh329