zhaojh329/oui #104: Automatic locale results in blank page for unimplemented language

j123b567, 2021-03-24:
When the default locale is automatic, but the client locale is not implemented (e.g. "cs"), client script fails and no page is displayed.

Deeper investigation:
`oui-ui-core/vue/src/App.vue` calls rpc function `oui.load_locales`

For unknown locale, `oui-httpd/files/rpc/oui.lua:load_locales` returns empty Lua object `{}`

This object is later identified as an object and not as an array by `oui-httpd/src/lua2json.c:lua_test_array`, because it is empty.

RPC then returns JSON result `{}` instead of `[]`

In `oui-ui-core/vue/src/App.vue`, on result, method `forEach` is called, which does not exists for object type, but exists for array type. This causes `Uncaught (in promise) TypeError: t.forEach is not a function` javascript exception and client side javascript completely terminates.

I would like to fix it myself, but I don't know the best place to do it.
Doing the similar trick like https://github.com/mpx/lua-cjson/pull/25/commits/3b1429fabbaa6c2fc3d8c0f7c83b3810e530caf9 would be nice but my knowledge of Lua is limited to create a PR. Other solutions are probably easier, but ugly (test special case of `{}` in Vue, etc.)

zhaojh329, 2021-03-25:
This is the tricky question of whether to turn to "{ }" or "[ ]" for an empty Lua Table.

zhaojh329, 2021-04-30:
https://github.com/zhaojh329/oui/commit/1a8197c60ea38438f8fe8594a9bc92423c68acfa

closed 2021-04-30 by zhaojh329