zhaojh329/oui #105: SSL Certificates commonname does not replace %D

j123b567, 2021-03-24:
After some investigation of `/etc/init.d/oui-httpd` I found, that I need either `px5g` or `openssl` binary to generate selfsigned keys. After installing `openssl-utils`, init script generates keys, but commonname is `%D`, which is the default value from configuration in `/etc/config/oui-httpd`

Where it was intended that this placeholder will be replaced? Probably by domain name? Is it a specific feature of some of these binaries? Or it is just currently unimplemented?

zhaojh329, 2021-03-25:
https://github.com/zhaojh329/oui/blob/master/oui-httpd/files/oui-httpd.init#L35

j123b567, 2021-03-25:
Yes, but the `commonname=%D` so `"${commonname:-OpenWrt}"` leads to `"%D"` which is later passed to `openssl`/`px5g` which use it as text and does not replace it by nothing meaningful.

So, prior this command, there should be some replacement logic, which is currently unimplemented?
Just need to know, if it is a bug, misconfiguration, or it is just not implemented yet.

zhaojh329, 2022-07-21:
Vue3 + Nginx + Lua is coming...

closed 2022-07-21 by zhaojh329