zhaojh329/oui #109: Javascript & CSS file are not readable, seems they are binary file

ecomsolution, 2021-04-07:
Description:
javascript files and css files are not text files, I expect they are pure text files, but looks like they are binary files, is this the format issue, encoding issue, or there is way to convert them to text?

such as:
oui/applications/oui-app-cron/vue/dist/app.js
chunk-vendors.662c50cd.css

zhaojh329, 2021-04-07:
gzip

ecomsolution, 2021-04-07:
Thank you for replying my question!
how do we decompress the compressed js file? Do we manually decompress it before we compile the QUI project? or is there a way to decompress the gzipped javascript automatically when the web browser load the page? See the view source of the javascript on the oui.htl page below.
[image]

zhaojh329, 2021-04-07:
You can modify oui/oui-ui-core/vue/vue.config.js and oui/build-app/vue.config.js

ecomsolution, 2021-04-09:
I ran the Nginx and Apache on the Openwrt device, but it still shows the compressed content for CSS and JS. I am new to this, would you please provide additional information on how to configure this? Thank you!

zhaojh329, 2022-07-21:
Vue3 + Nginx + Lua is coming...