zhaojh329/oui #134: 最新版openwrt默认启用了acl，请适配

zhangxiaoliangzxl, 2021-06-28:
Environment: (put here arch, model, OpenWrt version)
OpenWrt SNAPSHOT, r17037-56bdb6bb97

Description:
Uncaught (in promise) {"req":{"jsonrpc":"2.0","id":1,"method":"call","params":["","ui","first_login",{}]},"error":{"code":-32000,"message":"Access denied"}}
Uncaught (in promise) {"req":{"jsonrpc":"2.0","id":2,"method":"call","params":["","ui","get_lang",{}]},"error":{"code":-32000,"message":"Access denied"}}

最新版openwrt默认启用了acl，请适配一下

nickberry17, 2021-07-16:
I have the same problem. First time using the project. I am on a fresh OpenWrt installation which does not have LuCI.

[image]

zhaojh329, 2022-07-21:
Vue3 + Nginx + Lua is coming...