zhaojh329/oui #153: 打开是空白网页

shicetu, 2021-11-16:
环境: OpenWrt version 19.07

描述:
我打开对应网页的时候，是空白页面，然后F12有报错，这个报错我不怎么看的懂，我不会Vue
可以帮忙解答一些，这个问题是什么原因造成的吗？
[image]

shicetu, 2021-11-16:
oui-httpd -vv --home /www --rpc /usr/lib/oui-httpd/rpc -a (ip_addr):80 --db /etc/oui-httpd/oh.db --no-auth-file /var/run/oui-httpd-no-auth.js
on -w 2 &
使用这个命令，修改ip地址，在后台启动服务就好了

closed 2021-11-16 by shicetu

gustavoquincy, 2022-12-13:
也遇到了相同的问题，请问一下没有oui-httpd命令要怎么解决？

shicetu, 2022-12-13:
我当时是看了其他issue解决的这个问题，你看看其他的issue呢，我有点忘了当时怎么改的了-------- 原始邮件 --------发件人： Kaile Yuan ***@***.***>日期： 2022年12月13日周二 18:13收件人： zhaojh329/oui ***@***.***>抄送： shicetu ***@***.***>, State change ***@***.***>主    题： Re: [zhaojh329/oui] 打开是空白网页 (Issue #153)
也遇到了相同的问题，请问一下没有oui-httpd命令要怎么解决？