zhaojh329/oui #155: OUI not work

RadioOperator, 2022-01-08:
Environment: (put here arch, model, OpenWrt version):  Xiaomi 4C
Description:
I have tried many times according to the instrucation, still cannot open OUI login page, openwrt is working in background.  
My build is compiling pass, I do not know what packages missing or cannot selected "y" which makes me going wrong.  
Attached is my config file, could you help to check it?  or share your working config file (any hardwares) here, thanks.
[xiaomi4c.config.zip](https://github.com/zhaojh329/oui/files/7832608/xiaomi4c.config.zip)

RadioOperator, 2022-01-24:
Today, updated source and make it again, no errors, but still no OUI login page out, CLI works.
Please help me, send me your config file, thanks.

RadioOperator, 2022-01-24:
真的好郁闷，我的make通过，就是用不了。 到底怎么配置才能用呢？ 全网都找不到可以用的。

RadioOperator, 2022-01-24:
哪位好心人帮帮忙，给我一个你用的config文件，我自己寻找问题，谢谢。

RadioOperator, 2022-01-24:
我看到好多问题都没有人回答。
哪位好心人帮帮忙，给我一个你用的config文件，我自己寻找问题，谢谢。

RadioOperator, 2022-01-24:
我的config文件在一楼，我估计是我选择的项目不对，不知是多了还是少了。

RadioOperator, 2022-01-25:
我的要求不高吧。
哪位好心人帮帮忙，给我一个你用的config文件，我自己寻找问题，谢谢。

RadioOperator, 2022-01-25:
681个星星中有一个是我的，156个Fork，33个Watch的，问个问题没有人理睬，这是个问题。

RadioOperator, 2022-01-26:
？？？

RadioOperator, 2022-01-26:
又试了一次，还是一样不能用，我用的是最新的版本v21.0.1，是不是这个原因呢，一定要用v19.7.x吗。

RadioOperator, 2022-01-26:
Hi all, I got it work on v21.02.1. My problem was started from openwrt official configs for a special device with luci, should start from the "clean" configs, then config it manually.

RadioOperator, 2022-01-27:
let me share my config file here
[config.zip](https://github.com/zhaojh329/oui/files/7947369/config.zip)
.

RadioOperator, 2022-01-27:
多谢赵大侠的奉献，我试用了一个小时，谈谈我的感受。先说一下，我没有什么目的，只是对openwrt的软件开发感兴趣，全网搜索类似的UI替代品，希望找到比openwrt原装更好，更有潜力的。以下，随便聊聊我粗略的观感：

1. 既然是要发展成商用平台，就应该放弃openwrt的UI平台画面思维，重点是“实用，简洁，美观”的统一，专业的设置就交给openwrt原装吧，毕竟他已经太好了。如果做成模仿openwrt实现各种各样的功能，那还不如花时间改进openwrt，我知道赵大侠您已经为openwrt做了太多的贡献。这个是另辟蹊径，建议以某个喜闻乐见的商用路由器的UI画面做底，功能不多于商用功能即可，openwrt的缺点正是太太过于复杂，一般的技术人员都难以驾驭。
这里推荐一个GL.iNet路由器的UI思路，他是openwrt为底的，可以一键切换到openwrt的完整UI，一般用户可以使用其大大简化了的原厂UI，最常用的基本功能设置简单。他还带有一个厂家自有的远程遥控管理功能为客户服务，好象是免费的。
https://docs.gl-inet.com/en/3/setup/mini_router/first_time_setup/

2. 目前的OUI，我一开始用就遇到了大的问题，网络上没有完整的设置教程，虽然github的readme都是必需的，但是不够全面，失败的原因都是各种各样的踩坑。如果有几个常用路由器IC的完整的config文件可以给大家参考，相信大家就很容易上手。我上面的就是AR300M的，OUI搞好以后，我把它config的有关部分移植到我的小米路由器，很快小米也正常使用OUI了。

3.  目前的OUI，LAN正常，WAN正常。当我不使用WAN线，而是用WiFi客户端连接主路由上网（repeater）时遇到了麻烦，也许是我的config选项有问题，总之就是不行，最后只好将我在openwrt上面设置成功的network/wireless/firewall三个文件搬过来，总算可以用了。

4. 还是前面说过的，OUI的菜单项目太多了，重点是“实用，简洁，美观”的统一，也容易debug和维护。

谢谢

zhaojh329, 2022-07-21:
Vue3 + Nginx + Lua is coming...

closed 2022-07-21 by zhaojh329