zhaojh329/oui #175: oui-httpd does not compile

castillofrancodamian, 2022-05-16:
```
mkdir -p /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/tmp/.root-copy.flock -c 'cp -fpR /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-3.12.1/.pkgdir/libuhttpd-nossl/. /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libuhttpd-nossl_installed
echo "libuhttpd-nossl" >> /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/pkginfo/libuhttpd.nossl.install
make[3]: Leaving directory '/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/feeds/packages/libs/libuhttpd'
time: package/feeds/packages/libuhttpd/nossl/compile#0.19#0.07#0.52
make[3]: Entering directory '/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/feeds/oui/oui-httpd'
rm -f /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b/.configured_*
rm -f /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/stamp/.oui-httpd_installed
mkdir -p /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b
(cd /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b=oui-httpd-git-2022.124.19941-78c504b -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b=oui-httpd-git-2022.124.19941-78c504b -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/usr;/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b )
CMake Error at /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/host/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Libuhttpd: Found unsuitable version "3.12", but required is
  at least "3.13" (found
  /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/usr/lib/libuhttpd.so)
Call Stack (most recent call first):
  /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/host/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:592 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindLibuhttpd.cmake:27 (find_package_handle_standard_args)
  CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b/CMakeFiles/CMakeOutput.log".
make[3]: *** [Makefile:116: /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/build_dir/target-x86_64_musl/oui-httpd-git-2022.124.19941-78c504b/.configured_b0cff694c62270c4304602690d8e98b9] Error 1
make[3]: Leaving directory '/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/feeds/oui/oui-httpd'
time: package/feeds/oui/oui-httpd/compile#0.17#0.07#0.56
    ERROR: package/feeds/oui/oui-httpd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/oui/oui-httpd/compile] Error 1
make[2]: Leaving directory '/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt'
make[1]: *** [package/Makefile:110: /run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt'
make: *** [/run/media/franco/3efff9e6-250e-484e-8eac-39902e236edc/openwrt/include/toplevel.mk:230: world] Error 2

real    1m37,399s
user    1m9,785s
sys     0m13,708s
```

zhaojh329, 2022-05-16:
Updates libuhttpd to latest version.

castillofrancodamian, 2022-05-16:
I did a build from a clean source. Still not compiling.

IricBing, 2022-06-08:
我也遇到这个问题了，如何解决呢？

Paull, 2022-06-08:
oui needs higher version libuhttpd and sqlite3, they are in feeds/packages/

```bash
cat > /tmp/0001-oui-compile-patch-for-21.02.3.patch << EOT
From 603dc3d50881f6d4fdaf5d5cc24af3abfc43e743 Mon Sep 17 00:00:00 2001
From: Paull <paull@qq.com>
Date: Tue, 7 Jun 2022 04:27:52 +0800
Subject: [PATCH] oui compile patch for 21.02.3

---
 libs/libuhttpd/Makefile | 13 +++++++------
 libs/sqlite3/Makefile   | 12 +++++++-----
 2 files changed, 14 insertions(+), 11 deletions(-)

diff --git a/libs/libuhttpd/Makefile b/libs/libuhttpd/Makefile
index 072cf87a5..51442f07a 100644
--- a/libs/libuhttpd/Makefile
+++ b/libs/libuhttpd/Makefile
@@ -8,12 +8,13 @@
 include \$(TOPDIR)/rules.mk

 PKG_NAME:=libuhttpd
-PKG_VERSION:=3.8.0
+PKG_VERSION:=3.14.1
 PKG_RELEASE:=1

 PKG_SOURCE:=\$(PKG_NAME)-\$(PKG_VERSION).tar.gz
 PKG_SOURCE_URL=https://github.com/zhaojh329/libuhttpd/releases/download/v\$(PKG_VERSION)
-PKG_HASH:=cdf97020be8ef73e74f12e0703e0f871ebd26c641ce2cb31f67c90a79483c372
+PKG_HASH:=512d4abfced0ec916d985d3d1eac7528e6676eecae6aca27fdaf313374e29f16
+PKG_BUILD_DIR:=\$(BUILD_DIR)/\$(PKG_NAME)-\$(BUILD_VARIANT)/\$(PKG_NAME)-\$(PKG_VERSION)

 PKG_MAINTAINER:=Jianhui Zhao <zhaojh329@gmail.com>
 PKG_LICENSE:=MIT
@@ -42,13 +43,13 @@ Package/libuhttpd-mbedtls=\$(call Package/libuhttpd/Default,mbedtls,+PACKAGE_libu
 Package/libuhttpd-nossl=\$(call Package/libuhttpd/Default,nossl)

 ifeq (\$(BUILD_VARIANT),openssl)
-  CMAKE_OPTIONS += -DUHTTPD_USE_OPENSSL=ON
+  CMAKE_OPTIONS += -DUSE_OPENSSL=ON
 else ifeq (\$(BUILD_VARIANT),wolfssl)
-  CMAKE_OPTIONS += -DUHTTPD_USE_WOLFSSL=ON
+  CMAKE_OPTIONS += -DUSE_WOLFSSL=ON
 else ifeq (\$(BUILD_VARIANT),mbedtls)
-  CMAKE_OPTIONS += -DUHTTPD_USE_MBEDTLS=ON
+  CMAKE_OPTIONS += -DUSE_MBEDTLS=ON
 else
-  CMAKE_OPTIONS += -DUHTTPD_SSL_SUPPORT=OFF
+  CMAKE_OPTIONS += -DSSL_SUPPORT=OFF
 endif

 define Package/libuhttpd-\$(BUILD_VARIANT)/install
diff --git a/libs/sqlite3/Makefile b/libs/sqlite3/Makefile
index c55ec658d..81c2ce933 100644
--- a/libs/sqlite3/Makefile
+++ b/libs/sqlite3/Makefile
@@ -8,12 +8,12 @@
 include \$(TOPDIR)/rules.mk

 PKG_NAME:=sqlite
-PKG_VERSION:=3330000
-PKG_RELEASE:=1
+PKG_VERSION:=3370000
+PKG_RELEASE:=\$(AUTORELEASE)

 PKG_SOURCE:=\$(PKG_NAME)-autoconf-\$(PKG_VERSION).tar.gz
-PKG_HASH:=106a2c48c7f75a298a7557bcc0d5f4f454e5b43811cc738b7ca294d6956bbb15
-PKG_SOURCE_URL:=https://www.sqlite.org/2020/
+PKG_HASH:=731a4651d4d4b36fc7d21db586b2de4dd00af31fd54fb5a9a4b7f492057479f7
+PKG_SOURCE_URL:=https://www.sqlite.org/2021/

 PKG_LICENSE:=PUBLICDOMAIN
 PKG_LICENSE_FILES:=
@@ -31,6 +31,7 @@ PKG_FIXUP:=autoreconf
 PKG_INSTALL:=1

 PKG_CONFIG_DEPENDS := \
+	CONFIG_SQLITE3_BATCH_ATOMIC_WRITE \
 	CONFIG_SQLITE3_DYNAMIC_EXTENSIONS \
 	CONFIG_SQLITE3_FTS3 \
 	CONFIG_SQLITE3_FTS4 \
@@ -99,7 +100,8 @@ endef
 TARGET_CFLAGS += \
 	-DHAVE_ISNAN \
 	-DHAVE_MALLOC_USABLE_SIZE \
-	-DSQLITE_ENABLE_UNLOCK_NOTIFY
+	-DSQLITE_ENABLE_UNLOCK_NOTIFY \
+	\$(if \$(CONFIG_SQLITE3_BATCH_ATOMIC_WRITE),-DSQLITE_ENABLE_BATCH_ATOMIC_WRITE)

 CONFIGURE_ARGS += \
 	--disable-debug \
--
2.25.1
EOT
git -C feeds/packages/ apply /tmp/0001-oui-compile-patch-for-21.02.3.patch
```

castillofrancodamian, 2022-06-08:
And why not update it? It is supposed to be updated automatically or is already included.

castillofrancodamian, 2022-06-08:
> Updates libuhttpd to latest version.

And how should I update?

Paull, 2022-06-08:
> > Updates libuhttpd to latest version.
> 
> And how should I update?

replace the one in feeds/packages

cp feeds/oui/libuhttpd/Makefile feeds/packages/libs/libuhttpd/Makefile

castillofrancodamian, 2022-06-08:
> > > Updates libuhttpd to latest version.
> > 
> > 
> > And how should I update?
> 
> replace the one in feeds/packages
> 
> cp feeds/oui/libuhttpd/Makefile feeds/packages/libs/libuhttpd/Makefile

And isn't it better to update upstream?

Paull, 2022-06-13:
> > > > Updates libuhttpd to latest version.
> > > 
> > > 
> > > And how should I update?
> > 
> > 
> > replace the one in feeds/packages
> > cp feeds/oui/libuhttpd/Makefile feeds/packages/libs/libuhttpd/Makefile
> 
> And isn't it better to update upstream?

I don't know. Maybe the upstream version is good enough for non-oui users.

zhaojh329, 2022-07-21:
Vue3 + Nginx + Lua is coming...

closed 2022-07-21 by zhaojh329