zhaojh329/oui #194: error

lanhaner, 2022-09-02:
Environment: (x86_64, openwrt 21.02)

Description:
```
make[3]: *** [/var/openwrt/build_dir/target-i386_pentium4_musl/oui-ui-core-git-2022.244.05112-7f854ab/.prepared_9f1722a7e05cf59f6ddbe90f41788037_6664517399ebbbc92a37c5bb081b5c53] Error 127
make[3]: Leaving directory `/var/openwrt/feeds/oui/oui-ui-core'
time: package/feeds/oui/oui-ui-core/compile#0.07#0.02#0.13
    ERROR: package/feeds/oui/oui-ui-core failed to build.
make[2]: *** [package/feeds/oui/oui-ui-core/compile] Error 1
make[2]: Leaving directory `/var/openwrt'
make[1]: *** [/var/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/var/openwrt'
make: *** [world] Error 2
```

zhaojh329, 2022-09-02:
Please provied full log.

lanhaner, 2022-09-02:
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
make[1]: Entering directory `/var/openwrt'
make[2]: Entering directory `/var/openwrt'
make[3]: Entering directory `/var/openwrt/target/linux'
make[4]: Entering directory `/var/openwrt/target/linux/x86'
rm -f /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/localversion
/var/openwrt/scripts/kconfig.pl  + + /var/openwrt/target/linux/generic/config-5.4 /var/openwrt/target/linux/x86/config-5.4 /var/openwrt/target/linux/x86/generic/config-5.4 > /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /var/openwrt/.config >> /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.target
/var/openwrt/scripts/package-metadata.pl kconfig /var/openwrt/tmp/.packageinfo /var/openwrt/.config 5.4 > /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.override
/var/openwrt/scripts/kconfig.pl 'm+' '+' /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.target /dev/null /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.override > /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set
mv /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.old
grep -v INITRAMFS /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.old > /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set
rm -rf /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/modules
cmp -s /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.prev || { cp /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config; cp /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.prev; }
export MAKEFLAGS= ; [ -d /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/user_headers ] || make -C /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203 KCFLAGS="-fmacro-prefix-map=/var/openwrt/build_dir/target-i386_pentium4_musl=target-i386_pentium4_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/var/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="i486-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Sep  1 02:09:44 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/var/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/var/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.203 CC="i486-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/user_headers headers_install
grep '=[ym]' /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.config.set | LC_ALL=C sort | mkhash md5 > /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.vermagic
touch /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/.configured
rm -f /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/vmlinux /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203/System.map
make -C /var/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.4.203 KCFLAGS="-fmacro-prefix-map=/var/openwrt/build_dir/target-i386_pentium4_musl=target-i386_pentium4_musl -fno-calle [... cut ...]
make[3]: *** [/var/openwrt/build_dir/target-i386_pentium4_musl/oui-ui-core-git-2022.244.05112-7f854ab/.prepared_9f1722a7e05cf59f6ddbe90f41788037_6664517399ebbbc92a37c5bb081b5c53] Error 127
make[3]: Leaving directory `/var/openwrt/feeds/oui/oui-ui-core'
time: package/feeds/oui/oui-ui-core/compile#0.06#0.04#0.14
    ERROR: package/feeds/oui/oui-ui-core failed to build.
make[2]: *** [package/feeds/oui/oui-ui-core/compile] Error 1
make[2]: Leaving directory `/var/openwrt'
make[1]: *** [/var/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/var/openwrt'
make: *** [world] Error 2

zhaojh329, 2022-09-02:
[image]

Please read the [documentation](https://zhaojh329.github.io/oui/) carefully.

lanhaner, 2022-09-02:
Thanks. The error has been fixed.

closed 2022-09-02 by lanhaner