zhaojh329/oui #199: 编译后打开网页无法打开

jorislee, 2022-10-09:
Environment: (put here arch, model, OpenWrt version)
Openwrt 19.07
npm 8.19.2
node v16.17.1
Device Linux OpenWrt 4.14.275 #0 Wed Apr 20 10:38:55 2022 mips GNU/Linux
Chip: MT7628

Description:
1. 按照文档中对 Nginx 的补丁说明，修改了 ./feeds/packages/net/nginx/Config_ssl.in 与 ./feeds/packages/net/nginx/Makefile
2. 按照 文档中的说明，添加编译链接并编译完成。
3. 更新编译后的固件，打开 web 失败，显示 nginx 404
4. 修改 nginx.conf 添加 /www 目录，打开后显示空白网页。
5. 是否有更加详细的步骤指导以正确打开 web 网页，非常感谢。

zhaojh329, 2022-10-09:
ps 看下 nginx 有没有运行。

RadioOperator, 2022-10-09:
试一试清掉browser的缓存，好像要用https://....

jorislee, 2022-10-10:
@zhaojh329 
ps 查看是正常的 nginx: worker process 

默认编译的 nginx.conf 是不能正常打开网页，后面可以打开是仔细修改了这个文件，是否与这个 nginx 配置文件有关？

jorislee, 2022-10-10:
@RadioOperator 
清掉了缓存，并无变化，使用 https://.... 无法打开任何网页。
https 无法打开看起来像与 nginx 有关，是否被我遗漏了某些关键步骤？
当前是按照文档一步步操作的，如提示，将非常感谢。

RadioOperator, 2022-10-10:
我刚刚又试了一次，一开始因为192.168.1.1打开过我的ISP路由，当新的OUI安装以后，第一次打开192.168.1.1也是遭遇到404.

然后 我清浏览器缓存以后，就好了。
[image]

jorislee, 2022-10-10:
@RadioOperator 
您好，依旧是不行的，当前感觉是设备 nginx 没有编译进配置信息。
是否可以复制一份您的 /etc/nginx 的配置信息参考一下那，或者您编译成功的 bin 分享一下，做一些对比。
当前使用 19.07 版本+ MT7628 设备。

RadioOperator, 2022-10-10:
@jorislee 我也是按照作者的说明搞的，没有多余的操作，也没有改有关nginx的特别地方。
给你我的config file，你可以对照一下，我的cpu是ATH79的一种，MT7628的新版oui的我没有做过。
[oui-ar300m.zip](https://github.com/zhaojh329/oui/files/9744632/oui-ar300m.zip)

jorislee, 2022-10-10:
@RadioOperator 
感谢 您给的配置，从您提供的 .config 来看，您使用的 openwrt 版本较新。
而我当前使用的是 19.07版本，目前怀疑是版本相关。现在更新版本到 21.03 再进行测试一下。

closed 2022-10-12 by jorislee