zhaojh329/oui #202: OPENWRT编译OUI出错

HollyMadCow, 2022-11-08:
Environment: (put here arch, model, OpenWrt version)
OPENWRT最新拉取的版本
Description:
已经升级HOST主机NODE到19版
并使用HOST的NODE
```
Formating code blocks by wrapping them with pairs of ```
```

vite v3.1.6 building for production...
vite v3.1.6 building for production...
vite v3.1.6 building for production...
vite v3.1.6 building for production...
vite v3.1.6 building for production...
vite v3.1.6 building for production...
vite v3.1.6 building for production...
vite v3.1.6 building for production...
✓ 0 modules transformed.
vite v3.1.6 building for production...
[vite:vue] Failed to resolve vue/compiler-sfc.
@vitejs/plugin-vue requires vue (>=3.2.25) to be present in the dependency tree.
✓ 0 modules transformed.
[vite:vue] Failed to resolve vue/compiler-sfc.
@vitejs/plugin-vue requires vue (>=3.2.25) to be present in the dependency tree.
error during build:
Error: Failed to resolve vue/compiler-sfc.
@vitejs/plugin-vue requires vue (>=3.2.25) to be present in the dependency tree.
    at resolveCompiler (/home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/@vitejs/plugin-vue/dist/index.cjs:23:11)
    at Object.buildStart (/home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/@vitejs/plugin-vue/dist/index.cjs:2602:46)
    at file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:22710:40
    at async Promise.all (index 4)
    at async PluginDriver.hookParallel (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:22638:9)
    at async file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:23688:13
    at async catchUnfinishedHookActions (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:23088:20)
    at async rollupInternal (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:23686:5)
    at async doBuild (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/vite/dist/node/chunks/dep-db16f19c.js:45815:24)
    at async build (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-backup-git-2022.306.07306-2a38bcb/htdoc/node_modules/vite/dist/node/chunks/dep-db16f19c.js:45662:16)
error during build:
Error: Failed to resolve vue/compiler-sfc.
@vitejs/plugin-vue requires vue (>=3.2.25) to be present in the dependency tree.
    at resolveCompiler (/home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/@vitejs/plugin-vue/dist/index.cjs:23:11)
    at Object.buildStart (/home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/@vitejs/plugin-vue/dist/index.cjs:2602:46)
    at file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:22710:40
    at async Promise.all (index 4)
    at async PluginDriver.hookParallel (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:22638:9)
    at async file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:23688:13
    at async catchUnfinishedHookActions (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:23088:20)
    at async rollupInternal (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/rollup/dist/es/shared/rollup.js:23686:5)
    at async doBuild (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/vite/dist/node/chunks/dep-db16f19c.js:45815:24)
    at async build (file:///home/zlwh2y/lede/build_dir/target-x86_64_musl/oui-app-acl-git-2022.306.07306-2a38bcb/htdoc/node_modules/vite/dist/node/chunks/dep-db16f19c.js:45662:16)
✓ 0 modules transformed.
[vite:vue] Failed to resolve vue/compiler-sfc.
@vitejs/plugin-vue requires vue (>=3.2.25) to be present in the dependency tree.
✓ 0 modules transformed.
[vite:vue] Failed to resolve vue/compiler-sfc.

HollyMadCow, 2022-11-08:
看起来像是VITE版本过低？如何升级？

zhaojh329, 2022-11-08:
我用的 `node` 是 `v16.17.0`，有可能是你的 node 版本比较高，要不你安装 16 版本的试下。

HollyMadCow, 2022-11-08:
切换NODE到16.18.1也是同样的问题

zhaojh329, 2022-11-08:
Node 16，Node 18，Node 19 都可以编译。

# Node 16
```bash
zjh@linux:~/work/openwrt$ node -v
v16.17.0
zjh@linux:~/work/openwrt$ make package/feeds/oui/oui-ui-core/compile V=s
make[2]: Entering directory '/home/zjh/work/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/zjh/work/openwrt/scripts/config'
make[1]: Entering directory '/home/zjh/work/openwrt'
make[2]: Entering directory '/home/zjh/work/openwrt/package/libs/toolchain'
...
make[2]: Entering directory '/home/zjh/work/openwrt/feeds/packages/net/nginx'
echo "nginx-ssl" >> /home/zjh/work/openwrt/staging_dir/target-x86_64_musl/pkginfo/nginx.ssl.install
make[2]: Leaving directory '/home/zjh/work/openwrt/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/ssl/compile#0.14#0.01#0.13
make[2]: Entering directory '/home/zjh/work/openwrt/feeds/oui/oui-rpc-core'
echo "oui-rpc-core" >> /home/zjh/work/openwrt/staging_dir/target-x86_64_musl/pkginfo/oui-rpc-core.default.install
make[2]: Leaving directory '/home/zjh/work/openwrt/feeds/oui/oui-rpc-core'
time: package/feeds/oui/oui-rpc-core/compile#0.08#0.01#0.08
make[2]: Entering directory '/home/zjh/work/openwrt/feeds/oui/oui-ui-core'
touch /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/.prepared_9ec3c207f2fe033e7ea9eb58046d4fda_6664517399ebbbc92a37c5bb081b5c53_check
cp -fpR ./htdoc /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce
echo "VITE_OUI_LOGIN_VIEW="login"" > /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/htdoc/.env.local
echo "VITE_OUI_LAYOUT_VIEW="layout"" >> /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/htdoc/.env.local
echo "VITE_OUI_HOME_VIEW="home"" >> /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/htdoc/.env.local
PATH=/usr/local/bin:/usr/bin npm --prefix /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/htdoc install

added 224 packages, and audited 225 packages in 12s

36 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
touch /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/.prepared_9ec3c207f2fe033e7ea9eb58046d4fda_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/.configured_*
rm -f /home/zjh/work/openwrt/staging_dir/target-x86_64_musl/stamp/.oui-ui-core_installed
(cd /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/./; if [ -x ./configure ]; then find /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/ -name config.guess | xargs -r chmod u+w; find /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/ -name config.guess | xargs -r -n1 cp --remove-destination /home/zjh/work/openwrt/scripts/config.guess; find /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/ -name config.sub | xargs -r chmod u+w; find /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/ -name config.sub | xargs -r -n1 cp --remove-destination /home/zjh/work/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce=oui-ui-core-git-2022.310.27079-437e2ce -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce=oui-ui-core-git-2022.310.27079-437e2ce -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce=oui-ui-core-git-2022.310.27079-437e2ce -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/zjh/work/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/zjh/work/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/zjh/work/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/zjh/work/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/zjh/work/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/us [... cut ...]
Packaged contents of /home/zjh/work/openwrt/build_dir/target-x86_64_musl/oui-ui-core-git-2022.310.27079-437e2ce/ipkg-all/oui-ui-core into /home/zjh/work/openwrt/bin/packages/x86_64/oui/oui-ui-core_git-2022.310.27079-437e2ce-1_all.ipk
echo "oui-ui-core" >> /home/zjh/work/openwrt/staging_dir/target-x86_64_musl/pkginfo/oui-ui-core.default.install
make[2]: Leaving directory '/home/zjh/work/openwrt/feeds/oui/oui-ui-core'
time: package/feeds/oui/oui-ui-core/compile#27.92#5.88#21.80
make[1]: Leaving directory '/home/zjh/work/openwrt'
zjh@linux:~/work/openwrt$ 
```

HollyMadCow, 2022-11-11:
重新拉取整个项目，全新编译就可以了，就是卡在luci-base上了

closed 2022-11-13 by zhaojh329