zhaojh329/oui #206: how to start developing?

mrmmg, 2022-12-04:
Environment: unubtu 20.04, openwrt latest on proxmox

Description:
Hi
I installed openwrt on my proxmox and have access to openwrt by ip.
Now I want to test your package in my virtual openwrt also I developing with vscode on unubtu 20.04 

I have a important question about running your repository where? in my local machine (Ubuntu 20.04) or my openwrt (proxmox)

please explain to me that how can I edit my codes in Ubuntu and show results in proxmox openwrt?

(sorry for my bad english)

zhaojh329, 2022-12-05:
https://zhaojh329.github.io/oui/guide/getting-started.html

mrmmg, 2022-12-10:
With openwrt docs for developer I solved my issue.

thanks to you

closed 2022-12-10 by mrmmg