zhaojh329/oui #207: oui-rpc not found

gholamrezaeipt, 2022-12-10:
Environment: v22.0.3

Description:

1. Download openwrt from git (git clone)
2. install packages and feeds
3. install oui (add to feed config and install)
4. in oui-ui-core/htdoc directory run `npm install` & `npm run dev`
5. open given link in browser
6. in browser developer tools > networks i saw a 404 error: `localhost:port/oui-rpc` not found

in vite.config.json

```
server: {
      proxy: {
        '/oui-rpc': {
          target: 'http://192.168.1.151',
          secure: false
        },
        '/oui-upload': {
          target: 'http://192.168.1.151',
          secure: false
        },
        '/oui-download': {
          target: 'http://192.168.1.151',
          secure: false
        }
      }
    }
```

and 192.168.1.151 is Openwrt that installed on VMWare
[image]

**Question: Which part is wrong for development?**

zhaojh329, 2022-12-10:
You need to install oui to your device first, then you can debug it.

zhaojh329, 2022-12-10:
Don't select uhttpd.

gholamrezaeipt, 2022-12-10:
> Don't select uhttpd.

How can i do not select uhttpd? which file exactly?

castillofrancodamian, 2022-12-13:
> > Don't select uhttpd.
> 
> How can i do not select uhttpd? which file exactly?

In the make menuconfig you can select which packages to install and uninstall.

closed 2022-12-15 by zhaojh329