zhaojh329/oui #210: Should we limit the node PATH env var ?

judecumt, 2023-03-01:
Description:
Should we limit the node PATH env var in node.mk ?
```
ifneq ($(CONFIG_OUI_USE_HOST_NODE),)
OUI_NODE_PATH := PATH=/usr/local/bin:/usr/bin:/bin
NODE := $(OUI_NODE_PATH) node
NPM := $(OUI_NODE_PATH) npm
$(eval $(call ColorInfo,Using Node.js from Host))
else
```
Since I'm using [nvm](https://github.com/nvm-sh/nvm) for node install management, node is not installed in system's PATH.

zhaojh329, 2023-03-01:
Maybe I can add a menu config to set the path.

zhaojh329, 2023-03-01:
This should be set a default PATH env. If there is a Node exists in OpenWrt and left the PATH empty, then oui will use node from OpenWrt.

zhaojh329, 2023-03-02:
```
config OUI_HOST_NODE_PATH
	string "PATH for host node search"
	default "/usr/local/bin:/usr/bin:/bin:$$\(NVM_BIN\)"
	help
	  This will set PATH env for host node search.
```

```
Checking Node.js for build oui...
NODE: PATH=/usr/local/bin:/usr/bin:/bin:/home/zjh/.nvm/versions/node/v14.18.3/bin node
Using Node.js from Host
Node.js version: 14.18
```

judecumt, 2023-03-02:
> This should be set a default PATH env. If there is a Node exists in OpenWrt and left the PATH empty, then oui will use node from OpenWrt.

How about patsubst openwrt's host bin dir from PATH? Since openwrt's host bin dir is fixed as `$(TOPDIR)/staging_dir/host/bin`,
so `$(patsubst %/staging_dir/host/bin:,,$(PATH))` should do the trick.

judecumt, 2023-03-02:
> This should be set a default PATH env. If there is a Node exists in OpenWrt and left the PATH empty, then oui will use node from OpenWrt.

You are right, I think openwrt's host bin dir can be handled by sed.
```
OUI_NODE_PATH := PATH=$(shell echo $(PATH) | sed -E 's|[^:]{1,}staging_dir/host/bin:||g')
```

zhaojh329, 2023-03-02:
> > This should be set a default PATH env. If there is a Node exists in OpenWrt and left the PATH empty, then oui will use node from OpenWrt.
> 
> You are right, I think openwrt's host bin dir can be handled by sed.
> 
> ```
> OUI_NODE_PATH := PATH=$(shell echo $(PATH) | sed -E 's|$(TOPDIR)/staging_dir/host/bin:||g')
> ```
> 
> Below is the PATH after:
> 
> ```
> PATH=/home/liuchao/sandbox/openwrt/openwrt-22.03.3/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin:/home/liuchao/sandbox/openwrt/openwrt-22.03.3/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin:/home/liuchao/local/bin:...
> ```
> 
> openwrt's host bin dir is gone.

If thus, there is no need to add CONFIG_OUI_HOST_NODE_PATH.

judecumt, 2023-03-02:
Since #214 is merged, I think now host node search will satisfy all needs, this issue can be closed.

judecumt, 2023-03-03:
@zhaojh329 Thanks for your refactor about this issue!
But in 06656f14225689fed42f70bd719313b1a49bf0fb, 
```
ifneq ($(CONFIG_OUI_USE_HOST_NODE),)
NODE_PATH := PATH=$(PATH)
else
```
oui will use node from OpenWrt.

judecumt, 2023-03-03:
> @zhaojh329 Thanks for your refactor about this issue! But in [06656f1](https://github.com/zhaojh329/oui/commit/06656f14225689fed42f70bd719313b1a49bf0fb),
> 
> ```
> ifneq ($(CONFIG_OUI_USE_HOST_NODE),)
> NODE_PATH := PATH=$(PATH)
> else
> ```
> 
> oui will use node from OpenWrt.

Oh, I just find out `STAGING_DIR_HOSTPKG:=$(TOPDIR)/staging_dir/hostpkg`, and openwrt's host node is in hostpkg/bin, so this is not a problem.

zhaojh329, 2023-03-03:
> Oh, I just find out `STAGING_DIR_HOSTPKG:=$(TOPDIR)/staging_dir/hostpkg`, and openwrt's host node is in hostpkg/bin, so this is not a problem.

STAGING_DIR_HOSTPKG not included in PATH on this stage, but it's inclued in PATH on Prepare and Compile.

judecumt, 2023-03-03:
Thanks for your reply!
I think this issue is resolved.

closed 2023-03-03 by judecumt