zhaojh329/oui #270: How to use Composition API?

diegonc, 2024-01-12:
Currently I'm migrating some views back to Options API to be able to use ```this.$oui.call```.
As ```this``` is not available in Composition API (AFAIK), what would be the best way to issue an RPC call using ```$oui``` in that API?`

zhaojh329, 2024-01-14:
To use `$oui`, you can only use the options API.

diegonc, 2024-01-14:
I was having a look at the OUI plugin [here](https://github.com/zhaojh329/oui/blob/e1814e3c3087a00ac2c13a10b51245f77bfe4090/oui-ui-core/htdoc/src/oui/index.js#L270). It looks like the following diff could enable Composition API users to ```inject('$oui')``` in order to get access to the ```$oui``` plugin.

```diff
   install(app) {
     app.config.globalProperties.$oui = this
     app.config.globalProperties.$md5 = md5
+    app.provide('$oui', this)
   }
```

I'm missing something? Is that an acceptable aproach?

zhaojh329, 2024-01-15:
It seems to work well.

diegonc, 2024-02-08:
I just happened to test this on one of my new custom applications and effectively it works well.
So I went ahead and made a PR 😅

PR #286 (2024-02-09): oui-ui-core: allow injecting "$oui" in apps
When Oui instance is installed as a Vue plugin, call `app.provide` to allow Composition API based applications to use `inject` to get a reference to the plugin.

This was explored on issue #270 and seems to work well.

Closes #270
closed 2024-02-09 by zhaojh329