zhaojh329/oui #314: missing 'lua-cjson-lua5.3' package

RadioOperator, 2025-02-06:
Hi, when I try to [make menuconfig], I got missing 'lua-cjson-lua5.3' package:

`WARNING: Makefile 'package/feeds/oui/oui-rpc-core/Makefile' has a dependency on 'lua-cjson-lua5.3', which does not exist`

and then, compiling stopped with same error at 'package/install' stage.

```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency lua-cjson-lua5.3 for oui-rpc-core
 * pkg_hash_fetch_best_installation_candidate: Packages for oui-rpc-core found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for oui-app-acl:
 * 	lua-cjson-lua5.3
 * opkg_install_cmd: Cannot install package oui-app-acl.
make[1]: *** [package/Makefile:71: package/install] Error 255
```

Please help and thanks.

zhaojh329, 2025-02-07:
24.10 or main

RadioOperator, 2025-02-08:
> 24.10 or main

all ok, thanks.

closed 2025-02-08 by RadioOperator