zhaojh329/oui #323: oui-rpc returns 404 and oui-httpd fails with missing cjson module

yasalish, 2025-05-20:
Hi, I’m migrating from LuCI to Oui for a modern Vue interface, but I encountered several critical issues:

 1. oui-rpc 404 Not Found
POST https://192.168.0.10/oui-rpc → 404 Not Found
•	 oui-rpc-core is installed
•	 All required lua-eco-* packages installed
•	 rpcd is running
•	 ubus list | grep oui shows nothing
•	 ubus call oui.rpc hello '{}' → Command failed: Not found

 2. oui-httpd fails due to missing cjson:
/usr/sbin/oui-httpd:13: module 'cjson' not found:
  no file '/usr/local/lib/lua/5.4/cjson.so'
But:
opkg list-installed | grep cjson
lua-cjson - 2.1.0-r4
lua-cjson-lua5.3 - 2.1.0-r4
opkg files lua-cjson-lua5.3
/usr/local/lib/lua/5.3/cjson.so
It seems oui-httpd expects Lua 5.4 but the system is using Lua 5.3.

Questions:
1.	How do I get oui.rpc to appear in ubus?
2.	How should oui-httpd be configured to find the right Lua version?
3.	Is it possible to force Lua 5.3 for Oui components to fix compatibility?
•	OpenWrt built from source
•	Node.js 20.18 present
•	Oui components selected from menuconfig (oui-ui-core, oui-app-*, oui-rpc-core)
•	All lua-eco-* IPKs installed and confirmed

Any help is appreciated — I’ve spent days trying to set this up. Migration from LuCI would be much easier with better docs and clear instructions for getting the backend working.
Thanks

zhaojh329, 2025-05-21:
https://github.com/openwrt/packages/pull/26461

zhaojh329, 2025-05-21:
In addition, the uci also needs to be replaced with the lua5.4 version. The latest version of lua-eco integrates uci.
I will modify the oui after the merge of https://github.com/openwrt/packages/pull/26461.

zhaojh329, 2025-05-23:
https://github.com/openwrt/packages/pull/26562

yasalish, 2025-05-23:
RROR: unable to select packages:
  lua-eco-uci (no such package):
    required by: oui-rpc-core-25.142.56308~f8cea0d-r1[lua-eco-uci]
make[2]: *** [package/Makefile:99: package/install] Error 14

zhaojh329, 2025-05-24:
Wait this to be merged.

closed 2025-05-25 by zhaojh329